JohanBendz/com.philio #16: TypeError: Cannot read property 'toString' of undefined

Inversion-NL, 2017-02-17:
A crash occurred for your app (version 0.2.2) at Feb 17, 2017 5:49 PM (UTC).

Homey software version: 1.1.9

Crash report: 
    TypeError: Cannot read property 'toString' of undefined
    at Object.module.exports.ZwaveDriver.capabilities.my_gateway_control.command_report_parser (/drivers/tagreader500/driver.js:182:58)
    at CommandClass.<anonymous> (/node_modules/homey-zwavedriver/lib/ZwaveDriver.js:533:45)
    at emitThree (events.js:102:20)
    at CommandClass.emit (events.js:175:7)
    at /homey-app/wireless/zwave.js:26:25
    at /homey-app/helpers/client.js:1:1077
    at Array.forEach (native)
    at process.<anonymous> (/homey-app/helpers/client.js:1:1046)
    at emitTwo (events.js:87:13)
    at process.emit (events.js:172:7)

caseda, 2017-05-20:
this issue is actually not of this app, but of your BeNext app :p (and also already fixed)
just letting you know. 😄

Inversion-NL, 2017-05-20:
Haha, stupid.

closed 2017-05-20 by Inversion-NL