JohanBendz/com.philio #21: Add support for PAT02-1B

djanssen1, 2017-05-24:
Hi there,

Can you add support for PAT02-1B sensor. It is a sensor for temperature and humidity and look like the multisensor series

https://www.robbshop.nl/2-in-1-temperatuur-en-luchtvochtigheids-sensor-z-wave-plus-philiotech?sqr=philio%20temperat&

MbenzNo, 2017-05-27:
I would also like to have support for the PAT02-B sensor. If I try to include it as a generic Z-Wave device it is identified as a slave, exchanges security keys , command classes and so on, but in the end it stops at retrive wake up interval, and ends with the message : No Z-Wave device has been found. There does not appear an unknown device icon, but if I try to include it again it fails due to the device is included. I have to remove it from settings.

cscottmcqueen, 2017-07-14:
Also would like support added for this device.

TedTolboom, 2017-07-14:
See Pull Request #24

cscottmcqueen, 2017-07-17:
Just to confirm PAT02-1B tested and working great in v0.0.20

Many thanks

djanssen1, 2017-07-18:
I already tested and worked with Ted, Ted thanks! I'm closing this one..

closed 2017-07-18 by djanssen1

grolsch1986, 2019-01-08:
the temper alarm is always on whats the problem off this????