JohanBendz/com.philio #25: PAN08-support?

cbho, 2017-09-16:
Asked at the app store, but not sure if it's read:
What are the perspectives on adding support for the PAN08-roller shutter?
Regards,
Chr.

Inversion-NL, 2017-09-16:
Since I don't have the device I'm not implementing it, sorry.
GitHub pull requests are more than welcome though!

Drops70, 2017-11-23:
I am also interested in the support for the PAN08 roller shutter as I hava already seven of These installed in my home.
If a device is needed I can send you one for implementation.

Drops70, 2017-11-23:
Or you might only need the three IDs ?
Man-ID: 316 
Product Type ID: 1 
Product ID: 6

Inversion-NL, 2017-11-23:
Please create an pull request. I'll be more than happy to merge it.
I don't have much time, sorry

Drops70, 2017-11-24:
Thanks for Feedback. Opened a pull request. Hope I did it right as I am totaly new to this ....

Inversion-NL, 2017-11-24:
Great work.
However there are no changes in your commit

Drops70, 2017-11-24:
This was the only way to open a pull request. Honestly I am a totaly Newbie to github and programming.
I will look deeper into it, but I need help to get the PAN08 device integrated.
Whaterever I can do to get this done I will take care of.

Drops70, 2017-11-24:
Question:
If I copy the content of a supported device and adjust it like I think it will be OK, would you be able to check wether it is working or needs adjustments. If so, I will spend some time on the weekend to do so.

Inversion-NL, 2017-11-29:
You should be able to get away with changing the IDs and the capabilities.

Drops70, 2017-11-29:
Thanks. Just about to learn Java (will take time) and to get used to the development tools from Athom.
Think I will have a first try this weekend.

Drops70, 2017-12-03:
One additional question. Which files are required? Drivers, app.js, app.json, ..?

Drops70, 2018-01-04:
Hi, as it is merged now, how is the further process of testing? Will there be a new "beta" release so I can test the function of the PAN08?.
Sorry to ask, but I am still new to this......

Inversion-NL, 2018-01-04:
There weren't any changes in the commit so nothing has changed.

Drops70, 2018-01-04:
Sorry, but I dont understand.
Though this is what you needed ....

[image]

Drops70, 2018-01-08:
Hi, 
sorry don't want to bother you, but can you please reply as I have no clue how to proceed at the moment.
I realy need to get this device supported as I have to run my old controller only to get my blinds moving.

Many thanks!
Oliver

Inversion-NL, 2018-01-08:
The only change in this commit is one empty line.

[image]

Drops70, 2018-01-09:
Thanks for your Feedback.
You are looking at my first "commit" I had no clue what to do.
Please have a look one line after with the "Update PAN08" file.....
This Looks like this (and more):

[image]

Inversion-NL, 2018-01-09:
I'm sorry, I did not know.
Weird that GitHub doesn't show this at the pull request.

Inversion-NL, 2018-01-09:
Could you please up the version number in app.json and update the readme?
You can add yourself as a developer, if you'd like?

Drops70, 2018-01-09:
Sure, will do my best, but can take some time unti weekend.
Me a developer? Funny thing, but I might add myself as I did at least some work ... :-)

Drops70, 2018-01-10:
Did the job - at least hope so ...

Inversion-NL, 2018-01-11:
Yes, great work.
Did you run and test with this code?

Drops70, 2018-01-11:
Good morning. No I didn't do any test with the code. I would have done so knowing how to do it.
I know there is a way to do it with Homey , but I didn't look into this so far.

Inversion-NL, 2018-01-12:
Could you please test it?
It's in the developer documents on the Athom website

Drops70, 2018-01-16:
Yes I will test it. 
One question. Do I have to test my single code or is there already a "beat" version containing the PAN08 code. If so, which file do I have to download?

Drops70, 2018-01-17:
Hi,
I get closer. I've copied the driver from Github and installed it yesterday. Today I will try to include the PAN08 Controller.
As all other controllers have an own folder, could you please also open a folder for the PAN08 driver and maybe also copy the asset-folder from the PAN-06 into it (to use the same Image files).

Drops70, 2018-01-17:
Tested - not working :-(.
Will try to find the mistake ....

Drops70, 2018-02-12:
Hi, I have tested the driver by creating an own little app for Homey.
It is just a basic driver not supporting the whole features of the PAN08, but it works now since a few weeks :-). I will create a new pull request.

mhschaap, 2019-05-01:
Hi!
Thanks for your efforts! However, it does not work for me so far. I cannot add the PAN08 shutter controller, I get it added as "simple Z-wave device", which does not give me the desired functionality. Please see screenshot. Any hints? Could I submit a logfile or other details? Thanks!
[image]

caseda, 2019-05-04:
@mhschaap please share the ID's of the device you included.
you can find the ID's in the "simple" device's device settings, under "node information".
Then i can add them into the app.

mhschaap, 2019-05-04:
@caseda, Thanks for your reply, that would be great! I bought 4 devices, but have installed just one yet, as it did not work so far.  I use a different language, so I have added a screenshot which should include all ID's. Please let me know if you need more, or if I should run a test or create a logfile (although I have no experience with this). 
[image]

mhschaap, 2019-05-07:
> @mhschaap please share the ID's of the device you included.
> you can find the ID's in the "simple" device's device settings, under "node information".
> Then i can add them into the app.

@caseda thanks again for your reply! Do you require more info, then what I have added last weekend? Could I do a test-run. Please let me know. Many thanks!!!

caseda, 2019-05-07:
The id's are included in the app, the PR is awaiting acceptance of the developer, and then can be submitted into the appstore (beta channel).

But feel free to download the code and install the app (v0.1.1) via CLI

mhschaap, 2019-05-07:
it seems like Product ID is 21 for my device, and the code says "6". I am not a developer, but if I could simply edit the code on my Homey with Vi over ssh/telnet or such, I could give it a go? Please let me know. I have no experience with direct access to Homey. 
[image]

mhschaap, 2019-05-07:
> But feel free to download the code and install the app (v0.1.1) via CLI
Thanks again!! I will not be home today, but I will give it a try (have not done this before)

mhschaap, 2019-05-07:
> The id's are included in the app, the PR is awaiting acceptance of the developer, and then can be submitted into the appstore (beta channel).
> 
> But feel free to download the code and install the app (v0.1.1) via CLI

the "issues" has been resolved, but the updated code still lacks ID "21", while it was in your change-request. But I might not understand the github procedure

caseda, 2019-05-07:
you do need to install the beta app (which isn't in the app store yet) or the code via CLI, anything before v0.1.0 will not have the ID

mhschaap, 2019-05-07:
Thanks for your efforts, I appreciate. 👍 

I would have liked to have it working by now. My apparently supported devices can’t be used. But I have to deal with that

caseda, 2019-05-07:
Just wait only a few more days, don't think it will take long for the beta app be in the app store.
I know it can be frustrating.

mhschaap, 2019-05-10:
Just wanted to let you know that I have just downloaded via CLI version 0.1.1 and could successfully include my PAN08 device. I have not really used it yet, as I have very little time at this moment. Just wanted to say thanks!!

Few things I noticed:
- in an "if" condition I can choose "dim level has changed", literally "Dim-niveau is veranderd"
- in the Devices section (apparaten) it also shows the dim-level
I guess this is just a textual thing. Could post screenshots if you would like. 

I will keep you posted after proper testing. Thanks again

mhschaap, 2019-05-10:
[image]
I think there is a bug:
- if "state == up", then I can give the "up command"
- it should be if "state != up", then I should be able to give the "up command" 

Even when wrong wired, or such (which is not the case), this logic is reversed imho. I looked in the code of this App, but this logic is programmed at another level?

caseda, 2019-05-10:
It shouldn't be called dim level with that class, that's an athom bug.
Also yes, all those cards are created by athom's core functionality, don't have any control over that.

mhschaap, 2019-05-14:
I have a question about the intention of the "state up/down" booleans:

when sunscreen is 50% open (and shown as 50% open in the PAN08 device section):
- boolean up should be false and boolean down should be false as well right?
- we mean the full 0% or the full 100% and nothing in between?

Either way, without changing the position, when I run a test-flow, it gives different results for this query. That is anyhow strange. Any comments on that? 

[image]
[image]

In order to check against automatic control overriding manually opened sunscreens, it would be great to have a condition: "if sunscreen open > 5%" or such. If this condition is the case we know:
1. the sunscreen is down already, but manually set at whatever value; so we will not interfere.
2. we can check in case of wind or rain if the sunscreen is (partly) open, so that we can close it. But by this 5% check we avoid sending repeatedly up commands while it is up already

Thanks for sharing your thougths as well!

caseda, 2019-05-14:
You can do that with logic cards, if more then (tag percentage open): 5%

mhschaap, 2019-05-14:
> You can do that with logic cards, if more then (tag percentage open): 5%
> Further I have no clue how the rest is handled, I didn't implement the device, and never worked with roller shutter before with zwave.
> 
> I guess I should have deleted the capability anyway, and only leave the slider.

YESSS you are right, I did not recognize that! It is not available directly from the PAN08, but as a regular "logic card", perfect! This also solves the boolean test issues, I guess this is the more reliable approach!

mhschaap, 2019-05-14:
can you also explain the -100% <-> + 100% slider? what is considered -100% and what is +100%? 
normally I would expect: 0% is up, 100% is down, or vice-versa.
[image]

Thanks again!!

caseda, 2019-05-14:
It is relative, change from where it is now, is it at 50% change it -10% makes it 40% as new value

picaro1, 2019-06-24:
I'm new here.
Tried adding PAN08 to Homey. Version 0.0.30.
Got a generic Z-Wave device with only an 'power'-button. Not of any use.
Then de-installed the device.
Used the beta-version of the app and tried to add it again.
Now I've got the error message stating that the device is already added, however it does not appear in my list of devices!

caseda, 2019-06-24:
@picaro1 please try resetting your device to factory default, either by following the device manual, or from the developer tools -> z-wave -> remove device

picaro1, 2019-06-24:
Thanks @caseda !
Tried the factory default, that didn't work. However it worked through 'developer tools'.
Do you now how to calibrate the dim value?

caseda, 2019-06-24:
i do not have the device, so no

mhschaap, 2019-06-24:
Calibration is only needed by the device itself, not through Homey. It is described in the manual that comes with the product (press the button in certain way)

thestormhunter, 2020-02-07:
Hi everyone,

Does anyone know when the PAN-08 is included to Homey? Mine also is added as "simple Z-wave device". Or can someone please help to get it fixed? 

Thanks for your support!

mhschaap, 2020-02-07:
> Hi everyone,
> 
> Does anyone know when the PAN-08 is included to Homey? Mine also is added as "simple Z-wave device". Or can someone please help to get it fixed?
> 
> Thanks for your support!

Use the Beta version, there it’s included

caseda, 2020-02-08:
with the new app store there isn't really a beta version anymore, but a [test version](https://homey.app/nl-nl/app/com.philio/Philio/test/) <= click for the link

thestormhunter, 2020-02-08:
Thanks for the fast response. That one i had allready installed, i can select the PAN08 but it doesn't reconise the PAN08 that i have.

mhschaap, 2020-02-08:
> Thanks for the fast response. That one i had allready installed, i can select the PAN08 but it doesn't reconise the PAN08 that i have.

Please check if you really have v0.1.1 installed, this solved it for me

thestormhunter, 2020-02-10:
I have removed the app and installed it again. Now it works perfectly. Thanks for your help! Kind regards

closed 2020-06-06 by JohanBendz