JohanBendz/com.philio #27: Include PAN 16-1

stefanwitkamp, 2017-10-17:
Hi, would it be possible to include PAN16-1 based on the information from the Z-Wave Alliance and the implementation of PAN11 (which has the same functionality)? Information (including product ID):
https://products.z-wavealliance.org/products/2001

Inversion-NL, 2017-10-17:
Yes but I really don't have time to troubleshoot the issue if it isn't working in one go, sorry.

stefanwitkamp, 2017-10-17:
I see. Maybe we can try and sideload the new version to verify if it works with a know partner that has Homey + the plug?

Settings:

[image]

stefanwitkamp, 2017-10-17:
Ps, I understand this is a 'try in one go'. Let's see where we get.

jolny, 2017-10-26:
@Inversion-NL I've made a pull request for the PAN16-1, can you accept it and we'll see it it works? Thanks!
I was unable to test it this week unfortunately.

Inversion-NL, 2017-10-26:
I'm also unable to test since I don't have the device.
I'll await your test before merging your PR then.

jolny, 2017-10-26:
Hi again!
I've made another commit now, we've tested the plug and it works now :)
It needs a new icon (using PAN11 svg files at the moment), but other than
that everything seems OK.
So please accept the PR :D

Inversion-NL, 2017-10-26:
Great, thanks

closed 2017-10-26 by Inversion-NL