JohanBendz/com.philio #28: Adding support for Philio clones (or is it the other way around?)

LochNESS84, 2017-10-19:
Yesterday I downloaded the source for Philio to 'hack' support for the D-Link door sensor into the Philio app. As the Philio 3-in-1 and D-Link 3-in-1 are identical devices, I have to be honest it was nothing more than replacing the manufactorerID in the app.json. 

However, as D-link and Philio (and probably the Zipatto 3-in-1) all are identical at first sight. I would suggest adding these 'clones' into one app.

I'm not sure what every ID would be (I only have the D-Link 3-in-1 door/window sensor), but based on an educated guess, the other D-Link devices would share this ID. The philio app would in that case support more devices by simply adding manufactorer ID's to existing mapping.

One question, can you add multiple manufactorerID's into a single configuration? Eg:

`"zwave": {
				"manufacturerId": 108, 264, 316
				"productTypeId": 2,
				"productId": [
                    `

Inversion-NL, 2017-10-20:
I thought I read somewhere that it is not how it's supposed to work.
Every manufacturer should have its own app otherwise it could be confusing for users.

LochNESS84, 2017-10-20:
Seems like a valid point. It's just a shame as it takes more resources on Homey than would be necessary. 

Running two 'identical' apps because you have product-clone A AND product-clone B is a bit too much. However, based on what I read on z-wave alliance. Zipato and Philio do share their manufacturerId. So that is already one overlap solved.

caseda, 2018-01-11:
old(er) issue, but here is my input on this:
there are a lot of manufacturers that lookalike, for example:

if you look at the philio door/window sensor, there are *8* manufacturers with the same shell, even when glanzing over the manual,
but their inserts are only for 2 the same (not the philio one though), the rest is slightly different, either hardware wise, or settings/parameter wise, though most people can not tell the difference.

if you would add 8 manufacturer names in the app, because they look the same, it will become a mayhem for people that don't have both manufacturers.

this is the main reason for keeping different manufacturers apart, even if they look the same.

JohanBendz, 2020-06-06:
One app a brand is the way to go. Closing this.

closed 2020-06-06 by JohanBendz