JohanBendz/com.philio #29: Offset for temp/humidity?

zshh, 2017-10-19:
Any plans to add offset for temperature and humidity for PAT02-1B?
This sensors are not very accurate out of the box so offset possibility (+/-) for the temp/humidity values would be really handy.

Inversion-NL, 2017-10-20:
No, I'm using a driver from Athom and there isn't any functionalities like this in the driver.

zshh, 2017-10-20:
Hi,

Thank you for your reply. So only way to do this I guess is to ask Athom team for the possibility which I will do as soon as I figure out which section of their GitHub I should post it under.

Cheers

jolny, 2017-10-31:
Hi,
I thought this was an interesting feature so I talked to Athom about it They said the driver would not provide it as some devices has it built-in, but they gave me some pointers as to how it can be done on a per-device basis. Might give it a shot, I'll keep you updated.

Inversion-NL, 2017-11-01:
That would be great!

RogerGorissen, 2019-05-03:
Jolny, did you succeed in adding offset for humidity and temperature? If so, how did you do it? Regads, Roger

jolny, 2019-05-03:
Nope, sorry. I noted that it would be possible to do, but nothing more :)