JohanBendz/com.philio #32: missing_settings_key: rfOffCommand

Inversion-NL, 2017-11-07:
Error: missing_settings_key: rfOffCommand
    at changedKeysArr.forEach (/node_modules/homey-zwavedriver/lib/ZwaveDriver.js:228:51)
    at Array.forEach (<anonymous>)
    at ZwaveDriver.settings (/node_modules/homey-zwavedriver/lib/ZwaveDriver.js:213:18)
    at /manager/drivers.js:1:3981
    at ZwaveDriver.fs.existsSync.drivers.(anonymous function).ready (/manager/drivers.js:1:2404)
    at clientDriverMessage (/manager/drivers.js:1:2949)
    at Homey.ready (/manager/drivers.js:1:446)
    at Object.Homey.ready (/homey.js:1:1073)
    at /manager/drivers.js:1:400
    at /helpers/client.js:1:1040

Inversion-NL, 2017-11-07:
@jolny Do you know if you used a setting called rfOffCommand in your driver?

jolny, 2017-11-07:
Hi!
Probably this:

"RF_off_command": {
            "index": 9,
            "size": 1
        }

This parameter setting is available for PAN16.
Do you know what causes this error? Is the parameter missing from some other device or could it be because of something else?

Inversion-NL, 2017-11-07:
Might be the device was already added without this parameter?

jolny, 2017-11-07:
Checked the manual, this is what I could find:
PAN06-1x and PAN04-1x have these parameters , however PAN06-1 only shows two parameters in the manual and PAN04-1 shows 6.
However, these parameters were already added before so assume they were added by mistake. Are you attempting to add a PAN04-1 or PAN06-1, by any chance?

Inversion-NL, 2017-11-08:
No, haven't changed the code at all

jolny, 2017-11-08:
No I mean, were you trying to add a physical module (PAN04 or PAN06) to your Homey, or does this debug information pop up anyway? Maybe you have one of these modules already?

Inversion-NL, 2017-11-08:
I receive all crash reports from users.
This was a crash at a user, I don't own any of the pan series.

jolny, 2017-11-08:
I see!
If the app crashes completely, we should break up PAN04-1 and PAN06-1 into separate devices (with less parameter settings) and leave the modules I added with the current settings as they are the PAN04-1x/PAN06-1x ones. Unless there is some way to specify "optional" parameters?
But it's weird, these crashes should have happened before, too.

Inversion-NL, 2017-11-08:
Since the crash only happened once, I'd suggest we call it an incident and be done with it.
I haven't received any more crashes other than this one...

jolny, 2017-11-08:
It's possible someone tried to set one of the parameters on one of the devices that does not have them, so I agree. :)
Those devices are so old, I doubt there are that many still being used. If it pops up again, we can just split the devices.

Inversion-NL, 2018-12-28:
Agreed

closed 2018-12-28 by Inversion-NL