JohanBendz/com.philio #33: missing_settings_key: existenceEndpoint3

Inversion-NL, 2017-11-07:
Error: missing_settings_key: existenceEndpoint3
    at changedKeysArr.forEach (/node_modules/homey-zwavedriver/lib/ZwaveDriver.js:228:51)
    at Array.forEach (<anonymous>)
    at ZwaveDriver.settings (/node_modules/homey-zwavedriver/lib/ZwaveDriver.js:213:18)
    at /manager/drivers.js:1:3981
    at ZwaveDriver.fs.existsSync.drivers.(anonymous function).ready (/manager/drivers.js:1:2404)
    at clientDriverMessage (/manager/drivers.js:1:2949)
    at Homey.ready (/manager/drivers.js:1:446)
    at Object.Homey.ready (/homey.js:1:1073)
    at /manager/drivers.js:1:400
    at /helpers/client.js:1:1040

Inversion-NL, 2017-11-07:
@jolny Do you know if you used a setting called existenceEndpoint3 in your driver?

Inversion-NL, 2017-11-11:
A crash occurred for your app (version 0.0.22) at Nov 11, 2017 11:10 AM (UTC).

Homey software version: 1.5.3

Crash report: 
    Error: missing_settings_key: existenceEndpoint3
    at changedKeysArr.forEach (/node_modules/homey-zwavedriver/lib/ZwaveDriver.js:228:51)
    at Array.forEach (<anonymous>)
    at ZwaveDriver.settings (/node_modules/homey-zwavedriver/lib/ZwaveDriver.js:213:18)
    at /manager/drivers.js:1:3981
    at ZwaveDriver.fs.existsSync.drivers.(anonymous function).ready (/manager/drivers.js:1:2404)
    at clientDriverMessage (/manager/drivers.js:1:2949)
    at Homey.ready (/manager/drivers.js:1:446)
    at Object.Homey.ready (/homey.js:1:1073)
    at /manager/drivers.js:1:400
    at /helpers/client.js:1:1040

Inversion-NL, 2017-11-11:
Dear Robin,

A crash occurred for your app (version 0.0.22) at Nov 11, 2017 11:15 AM (UTC).

Homey software version: 1.5.3

Crash report: 
    Error: missing_settings_key: existenceEndpoint3
    at changedKeysArr.forEach (/node_modules/homey-zwavedriver/lib/ZwaveDriver.js:228:51)
    at Array.forEach (<anonymous>)
    at ZwaveDriver.settings (/node_modules/homey-zwavedriver/lib/ZwaveDriver.js:213:18)
    at /manager/drivers.js:1:3981
    at ZwaveDriver.fs.existsSync.drivers.(anonymous function).ready (/manager/drivers.js:1:2404)
    at clientDriverMessage (/manager/drivers.js:1:2949)
    at Homey.ready (/manager/drivers.js:1:446)
    at Object.Homey.ready (/homey.js:1:1073)
    at /manager/drivers.js:1:400
    at /helpers/client.js:1:1040

jolny, 2017-11-12:
OK, I'm guessing the only way to fix this permanently is to separate the PAN0x devices. Will have a look at it today, hopefully.

Inversion-NL, 2017-11-12:
I was thinking about this.
How can there be a key missing from a setting.
We shouldn't be able to save it without a key in the first place. What do you think about this?

I think we should first ask someone from Athom to have a look at it.

jolny, 2017-11-12:
Aah, are you @roadxy ? That makes sense.
Yes, agreed. Maybe they can help :)

Inversion-NL, 2017-11-12:
Yes I am ;-)

JohanBendz, 2020-06-06:
Fixed

closed 2020-06-06 by JohanBendz