JohanBendz/com.philio #34: Outdoor Motion Sensor PSP05 do not detect constant

MarcelS1988, 2017-11-23:
Hello,
My sensor doesn't detect the motion correct. When I move in front of the sensor is detect me but alfter 8 sec he stops and it will wait 8 sec before detect again.
See below my setting for what I can do better?
[image]

Inversion-NL, 2017-11-23:
I'm sorry but I don't know.
I only created the driver.

ArchiSpec, 2017-11-23:
Same query here. Having delved deeper into the sensor manuals, have tried manually changing configuration setting 09 to longer than 08 (as it states in the table on page 6 of the manual), but this still doesn't seem to work. 

Could you please add a parameter setting for config 9 (turn off light time) in the same way as you have provided a setting for config 8 (PIR re-detect interval time). To my mind, its no point providing one without the other. Quote from page 6 of manual (**_The Notice_** highlighted in bold under parameter 8 is exactly what I believe @MarcelS1988 and I are experiencing):
"

> Parameter 8 - PIR re-detect interval time:-
In the normal mode, after the PIR motion
detected, setting the re-detect time. 8
seconds per tick, default tick is 3 (24
seconds).
Setting the suitable value to prevent received
the trigger signal too frequently. Also can
save the battery energy.
**_Notice: If this value bigger than the
configuration setting NO. 9. There is a period
after the light turned off and the PIR not
start detecting._**

and 

> Parameter 9 - Turn Off Light Time
After turn on the lighting, setting the delay
time to turn off the lighting when the PIR
motion is not detected. 8 seconds per tick,
default tick is 4 (32 seconds).
0 means never send turn off light command.

So could you add that parameter 9 setting in the main tab please??

ArchiSpec, 2017-12-03:
@MarcelS1988 Did you try the manual configuration? I still can't get it to work. 
@Inversion-NL Any luck with the request? These motion detectors appear useless without that configuration ability. 
Thanks in advance.

MarcelS1988, 2017-12-04:
@ArchiSpec At the moment I didn't try anything more.
Wich configuration do you mean?
I think you are right and we need the parameter 9.

Inversion-NL, 2017-12-05:
I currently don't have time to look at it, sorry

MarcelS1988, 2017-12-06:
@Inversion-NL Do you know somebody else can help us with this issue? Or when do you think somebody will look at this issue?

MarcelS1988, 2018-01-11:
@caseda can you maybe help with this issue? It is still open.... Last time you was helping me great and I see you here also active. :)

caseda, 2018-01-11:
i think we people are miss interpreting this device from the start.
(sorry @Inversion-NL as it was my implementation anyway, did not know about this issue)

this is how the device works:
motion activates,
then the sensor falls asleep for 8 seconds (parameter 8) and will **not detect any motion** until this timeout is over, after the 8 seconds it will also turn off the motion alarm.
this is how philio implemented it in the device and is not homey's fault.
it just works a bit different then most z-wave motion sensors

if you want a longer "alarm on" time you will have to use the countdown app and let it trigger that.
[IF] alarm is turn on [THEN] set countdown timer to 30 seconds [AND THEN] turn lights on
[IF] timer runs out [THEN] turn light (or whatever) off

group 9 is only specific for association group 2 (and not 1 where homey is in), the "lighting group" created for direct association and is not needed for the workings in homey.

caseda, 2018-01-11:
was thinking a bit more about this one, and actually there is an easy way to build it into the app instead of being reliant on the countdown app, and thus added an extra cancellation setting for it (was like 15 minutes work as the principle for the tamper alarm is the same).

though i can't test this code if it fully functions until tomorrow evening (friday), maybe even later.
If someone can test it by installing it via CLI be my guessed:
https://github.com/caseda/com.philio/tree/add-motion-cancellation

in theorie it should work just fine.
no re-pairing is needed, _you do need to save the device's settings_ at least once if it was already included.
otherwise it will use the old way.

ArchiSpec, 2018-01-12:
@caseda Thanks for your insight! Where were you when we needed you? Ha ha!

Must admit never realised that Parameter 9 was only for Group 2 (my bad!) as that is only stated in a separate section of the manual! And consequently didn't realise that there was already a device parameter in the app! No wonder it wasn't doing anything when I manually set it in the Raw Config section!

Your solution with countdown timer works perfectly, THANK YOU. Sorry, unable to install via CLI & test your app solution, but if this works, even better.  Would this then be updated in the main app automatically once you are happy with it? 

Thanks again.

caseda, 2018-01-12:
Expert parameters overwrite what is in the normal settings, so that wouldn't matter, that parameter just wouldn't do anything in homey itself 😛.
Not automatically, will need to push the update to the owner, and he can then submit it to the app store

caseda, 2018-01-12:
just tested it myself, and works like a charm, so be prepared for an incoming update to this driver 😄

ArchiSpec, 2018-01-12:
Great news. Thanks for sorting. Look forward to the update.

PR #40 (2018-01-12): add custom motion cancellation + version bump
this will add a custom motion cancellation so the user is not dependend
on the countdown app for length of the motion alarm and not be reliant
on the blind time.
this wil fix #34
closed 2018-01-12 by Inversion-NL