JohanBendz/com.philio #36: Tamper Alarm - PST02-C

fredrikhanssonsk, 2017-12-04:
Tamper Alarm is always active and i havent found a way to null it.
Both tamper switches are pressed down.

CyberSponk, 2018-02-11:
Not resolved in Homey 1.5.6-RC11.

CyberSponk, 2018-02-14:
Not resolved in Homey 1.5.6-RC13.

Inversion-NL, 2018-02-16:
Most likely an problem in Homey https://github.com/athombv/homey/issues/2243 and https://github.com/athombv/homey/issues/1785

CyberSponk, 2018-03-09:
Not resolved in Homey 1.5.8-EXP from 08.03.18.

Inversion-NL, 2018-03-09:
Because it still isn't resolved in Homey: https://github.com/athombv/homey/issues/2267

CyberSponk, 2018-03-09:
Reported it Today in the linked Issue: **Not** fixed in 1.5.8-exp. Updated ~10 hours ago. WakeUp-Time of Devices 6 hours. Tamper-Alarm is On.

CyberSponk, 2018-05-14:
Not resolved in Homey 1.5.11-EXP.

CyberSponk, 2018-06-18:
Changed all Philio to Fibaro > Could not assist here in future.

ThomasHoussin, 2019-10-19:
I have the same problem (tamper alarm that don't go away) - using latest beta in the application. Is there a way to fix this ?

JohanBendz, 2020-06-06:
Hi @ThomasHoussin, can you please create a crash report from the app so that I can see what's wrong?

ThomasHoussin, 2020-06-23:
Done, hope it helps : 32e9080e-a8af-4102-a427-264d0ba54fa8

JohanBendz, 2020-11-21:
Hi @ThomasHoussin, can you please try this version and see if the issue is resolved? https://homey.app/a/com.philio/test/

closed 2023-01-29 by JohanBendz