JohanBendz/com.philio #43: Can't install from Athom Cli

SadhuRus, 2018-03-20:
When i install app by "athom app install" i have errors:

`✓ Validating app...
✘ Homey App did not validate against level `debug`:
[
    {
        "keyword": "enum",
        "dataPath": ".drivers[2].settings[0].type",
        "schemaPath": "#/items/oneOf/0/properties/type/enum",
        "params": {
            "allowedValues": [
                "text",
                "password",
                "textarea",
                "label"
            ]
        },
        "message": "should be equal to one of the allowed values"
    },
    {
        "keyword": "type",
        "dataPath": ".drivers[2].settings[0].id",
        "schemaPath": "#/items/oneOf/1/properties/id/type",
        "params": {
            "type": "string"
        },
        "message": "should be string"
    },
    {
        "keyword": "enum",
        "dataPath": ".drivers[2].settings[0].type",
        "schemaPath": "#/items/oneOf/2/properties/type/enum",
        "params": {
            "allowedValues": [
                "radio",
                "dropdown"
            ]
        },
        "message": "should be equal to one of the allowed values"
    },
    {
        "keyword": "enum",
        "dataPath": ".drivers[2].settings[0].type",
        "schemaPath": "#/items/oneOf/3/properties/type/enum",
        "params": {
            "allowedValues": [
                "checkbox"
            ]
        },
        "message": "should be equal to one of the allowed values"
    },
    {
        "keyword": "enum",
        "dataPath": ".drivers[2].settings[0].type",
        "schemaPath": "#/items/oneOf/4/properties/type/enum",
        "params": {
            "allowedValues": [
                "group"
            ]
        },
        "message": "should be equal to one of the allowed values"
    },
    {
        "keyword": "oneOf",
        "dataPath": ".drivers[2].settings[0]",
        "schemaPath": "#/items/oneOf",
        "params": {
            "passingSchemas": null
        },
        "message": "should match exactly one schema in oneOf"
    }
]
Not installing, please fix the validation issues first`

what is means?

Inversion-NL, 2018-03-20:
The new CLI has problems with SDKv1 apps.
You'd could report it to Athom.

caseda, 2018-12-28:
this issue should have been fixed in an homey update as it was missing an option in the list (number).

at least can't reproduce it anymore, could you try again when the PR is in?

JohanBendz, 2020-06-06:
Fixed in test version / v.1.0.1

closed 2020-06-06 by JohanBendz