JohanBendz/com.philio #44: PAN11-1D Receiving "Invalid command class" after including to Homey

pelleskrutt, 2018-04-03:
PAN11-1D Receiving "Invalid command class" after including to Homey 1.5.8 / com Philio 0.0.22.
Any ideas?
[logphilio.txt](https://github.com/Inversion-NL/com.philio/files/1873186/logphilio.txt)

taiks1, 2018-11-15:
Same problem with PAN11-1E. after including created device  invalid_commandClass_COMMAND_CLASS_METER. Homey Firmware Version: 1.5.12 / com Philio v0.0.22 - 03.11.17

caseda, 2018-12-28:
is still an issue for you guys?

if so can you give me the id numbers displayed in the device's settings (node information) when you include it when the app is disabled, so it is included as a "generic z-wave device".

taiks1, 2018-12-29:
Hi,

Yes still issue after installing it shows "Smart Energy Plug-In Switch
PAN11-1",
but when I try something, then gives error
invalid_commandClass_COMMAND_CLASS_METER
Noe information
Node ID 20
Secure Yes
Manufacturer ID 316
Product type ID 1
Product ID 17
Firmaware version 273
Device Class (Basic) BASIC_TYPE_ROUTING_SLAVE
Device Class (Generic) GENERIC_TYPE_SWITCH_BINARY
Device Class (Specific) SPECIFIC_TYPE_POWER_SWITCH_BINARY

BR,
Rain

Kontakt R Mengers (<notifications@github.com>) kirjutas kuupäeval R, 28.
detsember 2018 kell 14:35:

taiks1, 2018-12-29:
> have you tried to re-include the device? if you haven't please try it
> (close by homey).
>
> if you have tried that (close by homey), then we have a problem as the
> pan11 then will have different versions, ones with metering and once
> without..

Sorry, what to you mean "close by homey". I have tried remove device and
then include again This does'nt help.
Is there some other method?

Kontakt R Mengers (<notifications@github.com>) kirjutas kuupäeval L, 29.
detsember 2018 kell 23:28:

caseda, 2018-12-29:
close by, exactly as what it literally means, near by homey, so within 1 meter of homey, the closer the better.

illwillSE, 2019-01-04:
Hi!

I have a similar problem. I've got 2 PAN-11 1D switches without power-metering and they are registered as power-meter capable which results in the "invalid_commandClass"-error described above (they do not have power metering).

If I add the device with the philiotech app disabled (and thus as a basic zwave-device) it works fine for on/off.

My node informations is not exacly as taiks1 these are my findings:

Secure No
Battrery No
Manufacturer ID 316
Product Type ID 1
Produce ID 1
Firmware version (empty)
Device Class (Basic) BASIC_TYPE_ROUTING_SLAVE
Device Class (Generic) GENERIC_TYPE_SWITCH_BINARY
Device Class (Specific) SPECIFIC_TYPE_POWER_SWITCH_BINARY

On my old zwave-gateway the supplier had to do some magic to make it work because the devices  were reporting the wrong information to the gateway (I don't know what he did or what it reported) but he basically said that Philio Tech PAN-11 1D  was reporting the wrong device information to the gateway and he had to add a new device to make it work.

Frosth76, 2019-02-17:
I have exactly the same issues with PAN11-1D without power meter. I have 15 devices and not so eager to buy new.

claesg, 2019-10-29:
I had the same problem with my PAN11-1D. I switched to the beta version 0.1.1 of the Philio app and tried the 'Plug-In Switch PAN11-1D'. It now works as intended. I am very happy!

nicleo, 2020-05-30:
I have exactly the same issues with PAN11-1E.
Diagnostic report: 2933ac3f-0a40-40bd-a5bf-4bde196cfc59

JohanBendz, 2020-06-06:
Hi @nicleo, does it work if you switch to the test version and add it again?

nicleo, 2020-06-06:
> Hi @nicleo, does it work if you switch to the test version and add it again?

Sorry, I don't know what you mean with "switch to the test version".
I have v. 0.0.30 and "update automatically" enabled.
Keep in mind I have plenty of working Philio devices setup so I can't uninstall the app without destroying half of my home automation processes.

JohanBendz, 2020-06-06:
Hi again @nicleo.
The Test version is an early version of the upcoming release, it should not break anything you have working. Given the feedback from others I would say that the Test version works just fine (has not broken anything that used to work) for everyone that changed to it.

If the Test version does not work out you can revert to the Live version without having to repair anything. **BUT**, I do not have all the devices so I have not been able test if all devices work fine when switching between the versions. So at the end of the day it's up to you to decide if you think it's worth trying.

The test version can be found here: https://homey.app/sv-se/app/com.philio/Philio/test/

nicleo, 2020-06-06:
@JohanBendz 
Upgrade to test version 0.1.1 worked well. Everything still seem to work.
PAN11-1E still gives same error so I removed it and added again as PAN11-1. No change.

I removed it again and this time added as PAN11-1D and now I can control it without any error.
However, PAN11-1E has no power meetering so PAN11-1E needs its own device just like PAN11-1D (but without power meetering interface).

Thanks!

JohanBendz, 2020-06-06:
@nicleo 
Great to hear that upgrade went well. 👍 

Can you please create a crash report from the app, it would help me create a driver that fits the device.

nicleo, 2020-06-06:
Done!

e30f34ef-b67d-4c61-a0e9-1be9cc087352

Mvh,
Nick Leo

———————————————
NICnet
Dalby Viggeby
SE-755 91 Uppsala
Sweden

www.nicnet.se

sent from a NICnet mobile device

JohanBendz, 2020-06-07:
@nicleo Added to Test version 0.1.3, please try and give feedback

nicleo, 2020-06-07:
@JohanBendz 
Added PAN11-1E but it shows up as "PAN11-1" and still has a power interface.
[image]
[image]

JohanBendz, 2020-07-28:
OK, the issue with PAN11-1E was that the PAN11-1 still hold code that messed with PAN11-1E.
Now fixed, will work from next release.

If anything else still is not working please open a new issue.

closed 2020-07-28 by JohanBendz