JohanBendz/com.philio #46: Homey 2.0 can't insta

ff20000, 2018-12-19:
I want to install this app. But it won't let me.
I already reinstalled homey but this did not worked.

ff20000, 2018-12-19:
this is what it give back.

[ { "keyword": "type", "dataPath":
". ow.triggers[0].tokens[0].example", "schemaPath":
"#/de nitions/i18nObject/oneOf/0/type", "params": { "type":
"string" }, "message": "should be string" }, { "keyword": "type",
"dataPath": ". ow.triggers[0].tokens[0].example['en']",
"schemaPath":
"#/de nitions/i18nObject/oneOf/1/patternProperties/%5E.*%24/type",
"params": { "type": "string" }, "message": "should be string" }, {
"keyword": "oneOf", "dataPath":
". ow.triggers[0].tokens[0].example", "schemaPath":
"#/de nitions/i18nObject/oneOf", "params": {
"passingSchemas": null }, "message": "should match exactly
one schema in oneOf" }, { "keyword": "type", "dataPath":
". ow.triggers[0].tokens[0].example", "schemaPath":
"#/properties/tokens/items/properties/example/oneOf/1/type",
"params": { "type": "number" }, "message": "should be
number" }, { "keyword": "type", "dataPath":
". ow.triggers[0].tokens[0].example", "schemaPath":
"#/properties/tokens/items/properties/example/oneOf/2/type",
"params": { "type": "boolean" }, "message": "should be
boolean" }, { "keyword": "oneOf", "dataPath":
". ow.triggers[0].tokens[0].example", "schemaPath":
"#/properties/tokens/items/properties/example/oneOf",
"params": { "passingSchemas": null }, "message": "should
match exactly one schema in oneOf" } ]: [object Object]

caseda, 2018-12-28:
now writing an PR for this app fixing this issue, please stand by 😄

JohanBendz, 2020-06-06:
Fixed in test version / v1.0.1

closed 2020-06-06 by JohanBendz