JohanBendz/com.philio #47: PAN07-support?

MackanS, 2018-12-22:
I've got a PAN07. Please let me know if you have time to add this one. I'm willing to help. I've got a Telldus Tellstick ZNet Lite V2 where tje PAN07 is recognized. Maybe that can help you get the information about the unit you need to add it?

caseda, 2018-12-28:
can you specify the id's when it is added as generic z-wave device in homey? you can find them under the device's settings, under node information

MackanS, 2018-12-28:
Node ID 5
Manufacturer ID 316
Product Type ID 1
Product ID 5

caseda, 2019-05-05:
As the PAN07 is very similar to other devices already in the app, i've tried to add it into the SDKv2 rewrite of the app, it should work, it might not work fully.

MackanS, 2019-05-06:
How do I get the v2 rewrite version?

caseda, 2019-05-06:
you can either download the code, and install it via CLI

or wait until the app is in the beta channel of the app store, don't know how long this will take though.

JohanBendz, 2020-06-06:
Added to test / v.0.1.1

closed 2020-06-06 by JohanBendz