JohanBendz/com.philio #51: tamper alarm always on pat02-1b

grolsch1986, 2019-01-08:
the temper alarm is always on whats the problem off this

TechnoX, 2019-02-09:
I see the same problem on my PAT02-1B. Using Homey (model called early 2019 and version 2.0.2) with the homey app 2.0.6.250

JohanBendz, 2020-07-28:
Hi @grolsch1986 and @TechnoX 
Can any of you guys please restart the app, make sure you notice the issue described above and send me a crash report from the app?

TechnoX, 2020-07-31:
Hi Johan!

Finally updates here! Thanks!

How do I generate such report? The app has never crasched. Under settings and general I can create a diagnostics report. I got this ID:  919532A0D2

The problem persist though. See image below:

[image]

JohanBendz, 2020-11-21:
Hi @TechnoX, can you please try the test version here and see if your issue persists: https://homey.app/a/com.philio/test/

closed 2023-01-29 by JohanBendz