JohanBendz/com.philio #53: Homey v2. 0 pat02-1b kan ik niet meer toevoegen

grolsch1986, 2019-01-16:
Ik wil graag de pat02-1b opnieuw toevoegen maar homey geeft aan dat hij al toegevoegd is maar ik zie hem nergens tussen de apparaten staan

RogerGorissen, 2019-05-03:
Hoi Grolsch1986,

Ik heb hetzelfde met de Pat02-1b: eerste keer installeren lukte niet.
Bij de volgende keer installeren (3 x op kleine knopje aan de achterkant klikken) krijg ik de melding dat het apparaat al is toegevoegd. Maar ik zie het nergens staan!

Hoe heb jij dit opgelost?

Groet, Roger

caseda, 2019-05-04:
always remove/reset your device first before inclusion, you can find how in the device manual, or use the ...more -> settings -> zwave -> remove device, then activate the inclusion mode on your device

RogerGorissen, 2019-05-06:
My problem was solved after deleting the device in the settings - Z-wave part.

closed 2020-06-06 by JohanBendz