JohanBendz/com.philio #55: PAT02-1B shows wrong temperature units (F/C)

krotanSE, 2019-02-02:
When moving over my PAT02-1B from Zipato to Homey I encountered that I am unable to get them to show Celsius.
It shows the actual temperature in Fahrenheit in the app, despite me choosing Celsius in the settings for the unit. It also shows (C) as the unit, but the actual numbers are Fahrenheit.
Is this a know problem with the latest version or am I doing something wrong?
Running version 0.0.30

caseda, 2019-05-04:
it should have changed to Celsius in the device during inclusion, what homey version are you running

JohanBendz, 2020-06-06:
Hi @krotanSE, is this still an issue?

krotanSE, 2020-06-06:
Still an issue. Only way I’ve found to solve it is to run an exclusion and then inclusion again. Works 9/10 times. 
So I got all devices working and showing Celsius, but the root issue is still there.

JohanBendz, 2020-06-06:
OK, thanks @krotanSE. I can't see any bugs in the code for this and you are the only one with the issue so I will close the case. If you find this bug to be troublesome in the future please get back to me.

closed 2020-06-06 by JohanBendz