JohanBendz/com.philio #56: PST02-1B reports faulty Lux value

Ellghe, 2019-02-27:
Teh lux value reported by PST02-1B is far less than other light sensors???

caseda, 2019-05-05:
homey only receives values from the device itself, so if the values are wrong, it is fully the device itself, nothing the app can do about that.

Ellghe, 2019-05-06:
Homey report light value in Lux (lx),the same as Fibaro (I have 2) & Aeotec (I have 3). However does not PST02-1B (I have 2) report the light value in Lux like the other brands does.  Conlusion: Please do not present the light value as lx when it is not!

closed 2019-05-06 by Ellghe

caseda, 2019-05-06:
Aaah, I see, then it probably reports the brightness in percentage.(?) 
Then it needs a custom capability to make it percentage (which needs a re-pair), or see if I can make a work around to convert it into (approximate) lux. 
Device was added by a community member, so I didn't know.

I'll add it to my to-do list

caseda, 2019-05-06:
should be showing "correct" lux values in homey with version 0.1.1 of the app, the range of philio devices isn't big though, bare that in mind (maximum of 500 lux based on their specification)

Ellghe, 2019-05-06:
Perfect the range is no problems for me. When will 0.1.1 will be released?

caseda, 2019-05-06:
depends on the original developer of this app, and also Athom when they accept the app into beta.

if you can't wait and know how to install via CLI you can also download the code and install it that way