JohanBendz/com.philio #58: Unable to save advance setting in In Wall Dual relay PAN06-1 in switch B

guardianhell, 2019-03-21:
Hi, I am Having issue to save setting in Switch B, and the problem is when I switched on the Switch B, in Homey App the switch A look like turned on also, but the light isn't turning on (only in app view) so i tried to change "Selected Relay that can be controlled " --> Relay 2 only, but when i hit save i can't save the changes

caseda, 2019-05-04:
I can't see why it wouldn't work in the code, what do you mean with, can't save, as in the check mark is not clickable?, if so then one of the settings is out of bounds, please see if you can find any settings that are wrong

guardianhell, 2019-05-06:
Since homey updated their multichannel this issue has been fixed, but i still have one issue regarding Philio PAN-06, when i add PAN-06, there will be shown 3 button (Switch A-B, Switch A, Switch B), when i turn on switch B via Homey apps, in APPS view switch A look like turn on also, but only in APP perpective, the physical lamp still turn off, could you help to fix this one? thank you

caseda, 2019-05-06:
Could you see if this is fixed with the new SDKv2 rewrite (v0.1.0)? Should become beta pretty soon, or if you know how to install via CLI you can download the code and install it.

I don't know how these devices work precisely as I don't have them and they are implented by other members of the community

guardianhell, 2019-05-07:
Hi, i just tried v.0.1.1, the issues still appear as before switch b on, switch a also turning on in the apps only

caseda, 2019-05-07:
What happens if you turn switch a on/off? Does b also switch in the gui? Or physically too?

guardianhell, 2019-05-07:
if i turn on / off switch A, it doesn't effect to switch B. only switch B affected switch A

Thank you

caseda, 2019-05-07:
Hmmm, that's weird behaviour, one final thought, if you check the association group (settings of your device) is homey's id _only_ in group 1 like it is supposed to be? Or also on group 2 and 3.
If also in group 2 and 3, remove homey's id and save the settings and try again

guardianhell, 2019-05-07:
ok, default configuration associaton group 1,2,3 value is 1, and i tried to change group 2,3 value into " ", but the result is same

caseda, 2019-05-07:
Hmmm, then I'm not sure what it could be that causes this 🤔

liteautomation1, 2019-05-08:
Having the exact same issue. Turning on relay b of the modules shows the status of relay a as on, but it is not, just in the app. Turning on relay a does not affect relay b status at all. Happens on pan04 and pan06 dual modules. 
Also is there a way to hide or remove the switch to control both relays? This is not needed and it just adds more devices to the device list. I would rather create my own virtual switch that would control the relays or more. 
Also any chance of supporting the single module pan 03 ? 

Cheers. Good app.

caseda, 2019-05-08:
You can just delete the "both relay" device instance, the rest will keep working just fine.

As I don't have these devices it is pretty much impossible for me to fix this issue.

guardianhell, 2019-05-08:
hi liteautomation,

for your issue when turning on switch b, and switch a also affected not just in the app, you can go to advance setting and change setting on "selected Relay that can be controlled" from relay 1 and 2 into relay 1 only

liteautomation1, 2019-05-08:
@guardianhell, thanks but this isn't the issue. That option will allow that device to activate both relays at once. This is not our issue. When we activate relay b(2) on the module via the app, it switches just that relay on the module, and that load turns on, not both loads. But within the app, it show that relay a(1) is now active. But it is not, it is still off. It is some setting within the app that is triggering both relays to show them as active when only 1 is. As the same does not happen if you turn on relay a only.