JohanBendz/com.philio #59: Please the Dimmer PAD07

Ziggiz, 2019-04-30:
PAD07 IN-WALL DIMMER SWITCH
https://www.zwavetaiwan.com.tw/pad07

caseda, 2019-05-05:
As i don't have this device, and there are no other devices in this app that are very similar, i unfortunately can't add this device into the SDKv2 rewrite of the app.

And unfortunately i don't have enough time on my hands to let (other) people test a new implementation if i would add it into the app.

Maybe if i have time later on (way later, as in several months later), i could try to look at this.

If you are brave enough, and know how to write a bit of code, you could try to copy over a dimmer device from fibaro (FGD-212 driver) and adapt it to your device.

JohanBendz, 2020-06-06:
Hi @Ziggiz, are you able to add the device as a generic Z-wave device? If so, please provide the following information.

Device model/name:
Manufacture ID:
Product Type ID:
Product ID:

JohanBendz, 2020-06-07:
Added to Test version 0.1.3, please try and give feedback

closed 2020-07-28 by JohanBendz