JohanBendz/com.philio #60: Pat02-1B: status already added???

RogerGorissen, 2019-05-03:
The first install of my new Pat02-1B didn’t succeed. The next attempt, pressing the tamper key 3 times, results in the error message: status alteady installed. But I can’t find the device, so I van’t remove it. Uninstalling and reinstalling the app doesn’t help, restarting it neither.

How can I solve this? I can’t use the Pat02-1B now.

caseda, 2019-05-04:
please reset you **device** to factory defaults and try again

RogerGorissen, 2019-05-04:
Hello,

Thank you for your answer. I already succeeded by first removing the device from inside the Homey (more - settings -Z-wave - remove a device) and then again adding it from inside the Philio-app.

The device now works correct!!

Regards,

Roger

Verstuurd vanaf mijn iPhone

closed 2019-05-06 by RogerGorissen