JohanBendz/com.philio #62: PAT02-b not connected, no new data

RogerGorissen, 2019-05-06:
I've added the PAT02-B device saturday. Suddenly on sunday evening, about 19.00, no new data received by Homey. Device didn't refresh the data, even after changing the tick interval to 1 minute and the auto report humidity time to 1, so every minute new data has to be send.
Removed the device (struggle, followed correct procedure but device still was available in Homey, even after a few times) and added it again. Received data, everything ok.

Today the same thing happened: at about 19.45 no new data received. Testing it using http://developer.athom.com/tools/zwave the device is offline and unreachable.
The device has as one of its functions that a change in humidity of 5% will be reported immediatedly, despite of the timer settings. Strange enough, the data after a 5% change is received and reported by Homey, but still the device is offline and unreachable, according to Homey. 
Data isn't received every one minute, as I set.

Is this a fault in the Philio-ap, in the Z-Wave functionality of the Homey or is the device broke? Any other way I can test this, besides removing and adding it again?

caseda, 2019-05-06:
try putting the device closer to homey, or the other way around, this definitely sounds like a range thing

RogerGorissen, 2019-05-06:
Hi,

Tried it even with the device right next to the Homey. No change: offline, unreachable.
Removed and added it again. New data for a few times and then...nothing. No new data.

Is your experience that devices on a battery are reachable and online?

Can it be the case that the numbers of the parameters have been changed in the device and that the change in the parameters don’t affect it all? An alternative is to change the settings by sending RAW-data?

Do you still develop the Philio-app by the way?

Regards,

Roger

Verstuurd vanaf mijn iPhone

caseda, 2019-05-06:
Online is only possible when it is awake, 10 seconds maximum every time it wakes up, so 99% it is "offline" or in better words, asleep.
All my battery devices are reachable though, but my mesh network is pretty big and very stable.

I did not develop the app, philio is a collaboration of (too) many people, I only added 3 drivers (psp05, psr04 and psr07(new)), and did the sdkv2 rewrite just because I'm pretty knowledgeable with zwave, and wanted the app to keep working also in the future.

RogerGorissen, 2019-05-06:
Very strange, the unreachable status.

I’ve ordered a Fibaro power plug. The fan which is controlled by the Philio-device, is now connected by a COCO-device (433,92 MHz) and is located on the other side of the wall of the bathroom. Maybe with a Z-wave power plug the mesh coverage will improve.

At the same time I think something is wronf with either the device or the app (or the Homey?). A wired Fibaro dimmer is reachable, so the Homey seems to be able to talk to Z-Wave devices.

Will keep you updated.

Roger

Verstuurd vanaf mijn iPhone

closed 2020-06-06 by JohanBendz