JohanBendz/com.philio #65: device allready added.... eeeehhhh not visible at least

esebrani, 2019-11-07:
I have three philio  temp/hum sensors. everything has been working fine....
Now one sensor didn't report for a month, battery is ok so I decided to delete it and add it again.
Deletion was uneventful but when I was adding it again homey said "device already added"
Then I tried to delete the app and reinstalled it, reboot homey. Now are all three sensors unvisible but can't be added again....

I guess I need help here.....

mhschaap, 2019-11-20:
Had this also with my PAN08. I do not recall exactly, but I did use the full reset on the device, as per user manual of the PAN device

JohanBendz, 2020-06-06:
Hi @esebrani, what one needs to do in events like this is to enter the settings menu in the Homey app, click on Z-wave, and then choose to Remove a Zwave device (NOT reset network!!).

This can be done for ANY unit that you want out of Homey, that is stuck in Homey, that will not pair with Homey, that is brand new, that has been in any other controller.. etc.. remember that you need to push the pair button of the device while doing this, normally three times, sometimes five times, or even holding it for 10, 30 or more seconds. The device manual will tell.

closed 2020-06-06 by JohanBendz