JohanBendz/com.philio #68: PAN08 support

AmitPick, 2020-07-10:
I have installed test version 0.1.3 for PAN08 support and here is some feedback:
The Dim Level is not working properly. Two bugs are noted: 
1. Any dim level I select completely shuts down the roller
2. The dim level in the app always shows the roller is completely open. Even when I change the dim level and the roller shuts completely.

Perhaps what is also missing is the initial calibration similar to what Fibaro is doing.

JohanBendz, 2020-07-27:
Hi @AmitPick and thanks for the feedback.
I don't have the device myself so this information is vital to get for me!
I'd be extra happy if you could reproduce the problem and then send a crash report from the app. :)

AmitPick, 2022-10-05:
Hi @JohanBendz 

Sorry it took so long to reply back.
I am currently using v0.1.6.
I have reproduced the problem, but how do I send a crash report? The app has not crashed.