JohanBendz/com.philio #69: Flow trigger with Homey 5 broken?

per-lind, 2020-08-16:
Looks like I can no longer trigger flows on changes for PSR04. Same result with both dim changes and on/off. The value of the dimmer is however updated in the app so it looks like it is just affecting triggering of flows.
`
nodeVersion | "v12.16.1"
homeyVersion | "5.0.0-rc.23"
app version | 0.1.4
`

oilonboy, 2020-08-16:
I got the same issue with the same device. I can no longer dimmer my lights with the “Dim-level” tag since it’s not present as a choice anymore. 
Any solution to this?

per-lind, 2020-08-17:
I tried switching to https://homey.app/sv-se/app/com.philio/Philio/test/ and here it is working again.

JohanBendz, 2020-08-17:
Hi @per-lind and @oilonboy.
I'm sorry.. I thought I had published the new test version as live.. there was an error in the device trigger card in v.1.0.4 that I have fixed in v.1.0.6. Just sent a cert request to Athom to be able to update the test version to live.

oilonboy, 2020-08-30:
Hi again,
I still haven’t got the possibility to change the tag on the trigger card. It’s by default “level” and can’t be changed to the devices “Dim-level” as I had before the update. 

Please make it possible again.

JohanBendz, 2020-11-21:
Hi @oilonboy, sorry for my late answer. For some reason I had not noticed your comment here.

Are the other flow cards for the same device working for you as expected? Is it just that function that is not valid? Can you post a photo/screenshot of what you see? (I don't have the device myself so I can't test).

closed 2023-01-30 by JohanBendz