JohanBendz/com.philio #75: Connecting PST02-1 A

TommyGL, 2020-12-09:
Hey!
I have a problem connecting PST02-1 A multisensor 4/1. it's coming up.
"The device can no longer be reached. Try adding the device again by holding it closer to Homey."
Any tips on how I can solve this problem?

TommyGL, 2020-12-09:
Addition. yes I have already tried to keep it closer to Homey. //Tommy.

JohanBendz, 2020-12-09:
Hi there.
1. Try changing battery if you have not already.

2. Try pair it again, without removing it first. You should get a message telling you it's already paired. (if not you have to remove it and see if there is a "ghost" you also should remove first). Then restart the app. (or Homey).

Get back to me if this does not help.

TommyGL, 2020-12-10:
Thanks for quick reply! I have tested the following repeatedly and also in different sequences.
I have two units and both behave the same.

I have tried to change the battery.
I have also restarted both the app and Homey.
I have zeroed both devices.

What happens is that first Homey seems to detect the devices and then it thinks for a while and then it says "Wait while your device is added ..." then the second message comes up after about 30-40 seconds.

TommyGL, 2020-12-10:
I sent a diagonist report via Homey if it's helpful. 
Code
bf719bff-f7e6-46be-b559-9104dc5eb7e0

JohanBendz, 2020-12-10:
@TommyGL 
Sorry.. did not help much..

Stack trace
Note: this log has been manually submitted by a user.
Log ID: bf719bff-f7e6-46be-b559-9104dc5eb7e0

User Message:
TommyGL

Problem connecting PST02-1 A

---- stdout ----
2020-12-10 07:43:00 [log] [PhilioApp] com.philio running...

---- stderr ----

TommyGL, 2020-12-10:
I managed to connect the devices but they only report temp and light. I do not get the other values ​​in them. 
When I connected the device, I pressed the switch every 5 seconds until it was connected. Then Homey managed to find the device.