JohanBendz/com.philio #76: Add PSM02 4.1 problem

Peterv123, 2021-01-06:
On the Homey I want to re-use my PSM02 sensors. And I want to turn the Pir detection off.
Finally I get two off them working well.
But one is always not detecting the magnetic contact en always gives the motion detection alarm.
The temperature is readed just after the pairing proces and then it does not read anymore.

When I look in the history I see :
movement alarm: on 27 minutes ago
Contact alarm: on 28 minutes ago
Contact alarm: off 28 minutes ago

I disconnect and reconnect the sensor 20 times (the other two did it suddenly right) but still the same.

Two questions: 
Is it possible to pair the PSM02 in a reliable way?
How can you translate the bits mentioned by "operation mode" and "function switch" into integers?

JohanBendz, 2021-01-06:
Hi @Peterv123 

1. 
It should be, if you are able to pair it again please send me a crash report (use it a bit before sending so I can see events in the log).

2. 
Bit 0 is the one to the right and bit 7 is the one to the left (8 bits).
Change 0 to 1 to "set" the bit.

For the PSM02 this means Operation Mode is handled this way:
Bit0: 1 means security mode, 0 means home automation mode.
Bit1: 1 means enable test mode, 0 means disable test mode.
  Notice:
  Bit0 and bit1 will effect when the DIP Switch setting to program mode. If bit1 is enabled, the bit0 is useless.
Bit2: Disable the door/window function.
Bit3: Setting the temperature scale. 0: Fahrenheit, 1:Celsius
Bit4: Disable illumination report after event triggered.
Bit5: Disable temperature report after event triggered.
Bit 6 Not used (always 0)
Bit 7 Not used (always 0)

So a value for:
- home automation mode (bit 0 = 0)
- disable test mode (bit 1 = 0)
- enable door/window function (bit 2 = 0)
- Celsius (bit 3 = 1)
- enable illumination report (bit 4 = 0)
- enable temperature report (bit 5 = 0)
is 00001000, converted using the below url you get the value 8

https://www.rapidtables.com/convert/number/binary-to-decimal.html?x=00000000

Peterv123, 2021-01-07:
a929885e-6220-4d0c-8ff9-e93c2fb7e9cc

Hi @Peterv123

https://www.rapidtables.com/convert/number/binary-to-decimal.html?x=00000000

Peterv123, 2021-01-22:
did you receive the crash report? And have you found time to look at it yet?

Peterv123, 2021-02-01:
I have reset the sensors to factory default and now they will almost work in Homey.
The last thing to solve is the PIR part. In de Homey app the PIR alarm is always on.
I set the PIR sensitivity to 0 but the it still stay on.
Do you have a suggestion to solve this?

JohanBendz, 2021-02-10:
Hi @Peterv123, Sorry, been busy with other apps. Will have a look at this.

SortOfSmart, 2021-11-20:
Same here for the PST02. Unpaired and re-paired it four or five times, even restarted Homey for good measure. Nope. Every time it pairs, the very first motion alarm stays on and never goes off. The contact sensor works fine, and the others too, but I'm interested in the combined motion/contact sensor. If I can't get it to work right, I'll have to return it.

saintnick, 2022-09-11:
I second this here too. There's already another issue by someone on this. Would be great if we could acknowledge/clear the motion alarm.