JohanBendz/com.philio #78: Add PH-PAB01

jf84398942813, 2021-04-06:
Hi Johan,
First I would like to thank you for your job. 
I contact you because I want to replace my current Jeedom appliance by Homey and I use the Philio PH-PAB01 module to analyse my home power consumtion.
I see in your list this module is not yet supported.
Do you think this will be supported soon ?

Thanks in advance.
Best regards.

JohanBendz, 2021-04-16:
Hi @jf84398942813, thanks for your request, I'll have a look at it. Can you please add it as a generic Z-wave device and add the devices generic info to the request? (The Z-Wave Device Information in the device settings).

jf84398942813, 2021-04-19:
Hi Johan,
Below details :
----
Name : PH-PAB01.eu Micromodule Energy Meter
Fabricant : Philio Technology Corporation
----
Sûr : ⨯
Batterie : ⨯
ID de l'appareil : 189
ID du fabricant : 316
Product Type ID : 6
ID du produit : 26
Version du micrologiciel : 1
Sous-version du micrologiciel : 2
ID du micrologiciel : 1562
Version du matériel : 1
Classe d'appareil (Basique) : BASIC_TYPE_ROUTING_SLAVE
Classe d'appareil (Générique) : GENERIC_TYPE_METER
Classe d'appareil (Spécifique) : SPECIFIC_TYPE_WHOLE_HOME_METER_SIMPLE

Best regards.