JohanBendz/com.philio #79: Philio App for Siren PH-PSE02-1B can't play DoorChime

HakanDicander, 2021-04-26:
When creating a Flow to play the DoorChime sound on the Philio Siren an error flowcard_not_registered appears in Homey. The App should send a NOTIFICATION_REPORT command to the siren. Please see the attached PDF for details.
[PhiloSirenProblem.pdf](https://github.com/JohanBendz/com.philio/files/6379766/PhiloSirenProblem.pdf)

ReneVerwoerd, 2023-01-17:
I just bought the Philio PSE02 and want to use is as a DoorChime as well. 
@JohanBendz is it possible for you to fix the "flowcard_not_registered" error?
@HakanDicander Did you by any chance found a workaround to get the other sounds to work?

[image]

JohanBendz, 2023-01-30:
Hi @ReneVerwoerd.
Ooops. looks like I did some typos.. a new test version is on it's way but it is also a test version for the new Homey Pro - using SDK 3 and an updated Z-wave driver... so I really really need feedback on how it works... 
https://homey.app/a/com.philio/test/

ReneVerwoerd, 2023-01-30:
Thank you so much for updating! I will test it tomorrow, the kids a sleeping right now, not a good time to test a siren, haha! :-)

HakanDicander, 2023-01-30:
Hi, wonderful that there is some action regarding this!
In my understanding the only way to make the PSE02 to play different sounds is to send a notification command to the siren device, so please check if that is being done in the App. The slightest error or doing something else will result in no sound or in the default loud aram sound. I attach my original PDF that have info about how to do it.
I did find a way to work-around the problem, so in case it helps anyone of you, I share the Flow I start when I want my PSE02-1B play the dorr-bell sound.

Sorry for the Swedish, that is how my Homey show things, but I think you can understand.
There is a Z-Wave App, it has a Send card where you specify the bytes that will be sent as a Zwave-command to a given node id (see below). If you check at the end of the attached PDF you can see what values to you should put instead of the 0x06 and 0x16 to make it sound in other ways.
I use the Card from the Philio app to turn the siren off after 3 seconds, it works fine.
The node id to put in the Zwave Send card you find on the advanced settings for your Philio Device, here is how it looks in my Homey, I marked the node id with a red ring.

 If anyone of you have any more question about this I am happy to answer, I will however go skiing the rest of the week, so I can't try anything out until the earliest next week.
Best Regards
Håkan Dicander

ReneVerwoerd, 2023-01-31:
> Hi @ReneVerwoerd.
> Ooops. looks like I made some typos.. a new test version is on it's way but it is also a test version for the new Homey Pro - using SDK 3 and an updated Z-wave driver... so I really really need feedback on how it works...
> https://homey.app/a/com.philio/test/

Hi Johan, 

it still needs some TLC:
1. Install Siren works fine
2. Turning it on by ‘on/off’ button in app doesn’t work (“cloud not get device by id”)
3. Add flow card doesn’t work (keeps loading)

[image]
[image]

JohanBendz, 2023-01-31:
Hi again.
Please send a crash report form the app, mark it with your name/alias.

ReneVerwoerd, 2023-02-01:
I’ve made a diagnostic report and is says to give you this code: 754510ec-1a52-457a-80ad-7959dedb7e42
Is that what you need? Sorry, never made a crash report before

JohanBendz, 2023-02-01:
Thanks @ReneVerwoerd, you did great! :)

New version published, please try and let me know what happens. Send new diag report if you run into problems.

ReneVerwoerd, 2023-02-01:
@JohanBendz here you go:
1f74deab-7fd6-40a3-bf85-46cfffba7f56

[image]
[image]

JohanBendz, 2023-02-01:
Thx a bunch @ReneVerwoerd, new version out.

ReneVerwoerd, 2023-02-01:
@JohanBendz here you go:
e0c5cef5-3c6b-4d0f-ac09-74634ff0f9ff
[image]

JohanBendz, 2023-02-01:
@ReneVerwoerd
hahaha, closing in.. It's a pain in the **** not to have the devices one develop support for.. well well.. one more test version released.

ReneVerwoerd, 2023-02-02:
@JohanBendz 
For me it’s absolutely no problem to help with this fix! Just grateful that you spend time and effort to do this! 🙏

Next diagnosic report: abdafe15-eea3-40fe-8e99-d03f35bfc6b1
[image]

JohanBendz, 2023-02-02:
@ReneVerwoerd 
Thanks, I really appreciate your help!
New test version published.

ReneVerwoerd, 2023-02-02:
@JohanBendz 
Diag report: 9f577c29-9485-4ece-a416-d8b6af428df2

[image]

JohanBendz, 2023-02-02:
@ReneVerwoerd, oh.. darn.. typo... fixed! :)

ReneVerwoerd, 2023-02-02:
@JohanBendz next one: 
😉
5656f1a5-15fa-4f79-81f2-3d59e6d7d675
[image]

ReneVerwoerd, 2023-02-02:
@JohanBendz next one:
5656f1a5-15fa-4f79-81f2-3d59e6d7d675
[image]

JohanBendz, 2023-02-02:
@ReneVerwoerd
Oh.. no comment... lol... new version out..

ReneVerwoerd, 2023-02-02:
@JohanBendz The app works!! I can use diffrent chimes!
thank you very much for your effort! Much appriciated!

JohanBendz, 2023-02-02:
@ReneVerwoerd, great news 😊👍
Then I'll fix rest of the drivers just like this one.
Thanks you for enduring the massive testing 😂

closed 2023-02-02 by JohanBendz