JohanBendz/com.philio #86: Is the PHT_PSE04-EU supported in your app

henkbart, 2022-05-17:
Hello,

Just a question if the PHT_PSE04-EU outdoor sirene is supported in your app??
I am willing to buy this, but i need to know if i can use it with my Homey.

Thank you.

Henk Bartels

JMBijvank, 2023-07-24:
Hi Johan,

It would be super if you could add driver support for the PSE04. Do you need any info regarding the device which I could send to you?

tnx in advance.

CoffeMode, 2024-03-24:
Hello! How do you go about adding the PSE04 outdoor siren to the Homey app?

JMBijvank, 2024-03-26:
Hi,

Because the PSE04 can’t be selected I’m trying the PSE02, I also get to enter the pincode.  however the device ends up adding as a generic zwave.

However on my new Homey Pro it gives another error. Currently I'm at work so I'll try it this evening and send you the other error.

Kind regards,

Jeroen

https://www.int13h.nl/<http://cms.int13h.nl/>

Disclaimer:
By sending an email to ANY of my addresses you are agreeing that:

  1.  I am by definition, "the intended recipient"
  2.  All information in the email is mine to do with as I see fit and make such financial profit, political mileage, or good joke as it lends itself to. In particular, I may quote it on usenet.
  3.  I may take the contents as representing the views of your company.

This overrides any disclaimer or statement of confidentiality that may be included on your message.

JMBijvank, 2024-03-29:
A promised:

I’ve included the device and I get this overview:
[A screenshot of a phone  Description automatically generated]

When I click it, I get the following message:
[A red and black rectangle with white text  Description automatically generated]

This is the url which contains the user manual:
https://static1.squarespace.com/static/5c1afbd1266c07479bc41525/t/6221b675ee192a45634caf09/1646376568209/PSE04_UserManual_A3-20220303.pdf

Kind regards,
Jeroen

https://www.int13h.nl/<http://cms.int13h.nl/>

Disclaimer:
By sending an email to ANY of my addresses you are agreeing that:

  1.  I am by definition, "the intended recipient"
  2.  All information in the email is mine to do with as I see fit and make such financial profit, political mileage, or good joke as it lends itself to. In particular, I may quote it on usenet.
  3.  I may take the contents as representing the views of your company.
This overrides any disclaimer or statement of confidentiality that may be included on your message.

https://www.int13h.nl/<http://cms.int13h.nl/>