JohanBendz/com.philio #89: Update for Homey Pro 2023 coming?

Karpazu, 2022-11-02:
Hi,

Can you please update your app to be compatible with Homey Pro 2023.

Thanks, Karpazu

JohanBendz, 2022-11-02:
Hi @Karpazu.
Yes, I will update all apps  I manage to SDK3. This will make them support Homey Pro 2023.
I can't promise this app will ever support the bridge though as Athom have to approve pushing an app to the cloud.

Karpazu, 2022-11-03:
Hi!

Good to hear 👍

Thanks

Dici71, 2023-01-05:
Hi,

seems that first homeys will be shipped by mid of January...have you got a actual status on the Philio App Update ?

ReneVerwoerd, 2023-01-18:
> Hi @Karpazu. Yes, I will update all apps I manage to SDK3. This will make them support Homey Pro 2023. I can't promise this app will ever support the bridge though as Athom have to approve pushing an app to the cloud.

@JohanBendz When you update the app to SDK3, could you please also fix this: https://github.com/JohanBendz/com.philio/issues/79#issuecomment-1385338391

Many thanks in advance! :-)

JohanBendz, 2023-01-30:
See https://github.com/JohanBendz/com.philio/issues/91

closed 2023-01-30 by JohanBendz